openego/ding0 #223: Correct MV transformer parameters

gplssm, 2017-08-19:
Parameters of MV transformers are in incorrect unit (s_nom) respectively are missing (x,r). 3 TODOs

- [ ] Provide s in kVA
- [ ] Add r to type
- [ ] Add x to type
- [ ] Afterwards: edit PyPSA interface in eDisGo accordingly, see [eDisGo issue](https://github.com/openego/eDisGo/issues/19)

nesnoj, 2017-08-21:
A good source is pandapower, see its [doc](https://pandapower.readthedocs.io/en/v1.4.0/std_types/basic.html#three-winding-transformers). (pyPSA is also [using](https://www.pypsa.org/doc/components.html#transformer-types) it).
But I cannot find r and x values, I guess they cannot be calculated from the given params? @jochenbuehler

gplssm, 2017-08-21:
I already discussed about transformer parameters on the PyPSA mailing list. In [the discussion thread](https://groups.google.com/forum/#!topic/pypsa/6L3N6kNH2VY) you find the formula to calculate R and X

nesnoj, 2017-08-22:
@jochenbuehler Please check your inbox.
The pyPSA/pp types got different nominal ratings and they list only 3 types.
One more thing: They have different types for 10 and 20kV (with significant differences in R and X for the same voltage!).
@jochenbuehler Should we replace our types and account for voltage?

mltja, 2023-03-03:
Seems obsolete or fixed reopen if needed.

closed 2023-03-03 by mltja