openego/ding0 #224: implement constraint on max. half-ring length

nesnoj, 2017-08-21:
Currently, max. ring length is set to 60km, this might result in half-ring lengths of >30km.
Add max. half-ring length of 28km as constraint (30km according to [FGH](https://github.com/openego/ding0/issues/201#issuecomment-318044762)) - 2km for branching stubs.

nesnoj, 2017-08-21:
Solved in 4fe5124f3a61df753b71500b2ce0542741a0e81f.
However, some MVGD show longer half-rings than 28/30km (e.g. 3093) since this constraint is taken into account during improvements only and not in the savings method. This is how it is supposed to work because otherwise, very far nodes could not be connected at all.
This new method will have a large impact on results.

@gplssm: please generate a new dataset (we could talk about it tomorrow in natura). Please notify when it is done.
@abokam, @miguelparadacontzen: as soon as the new data has been generated, you may update your statistics.

gplssm, 2017-08-21:
Yep, I can do that!

nesnoj, 2017-08-24:
@gplssm **Important:** Please close CircuitBreakers prior to export, otherwise the lines to the CBs won't be exported! (see also https://github.com/openego/eDisGo/issues/23)

gplssm, 2017-08-28:
I included closing of CB's for `example_multiple_grid_districts.py` in 6a79d84003aff15af70b9214ee5fcf517377d286.

Would it be wise to include this as 13th step of `run_dingo()` rather than doing it at multiple locations?

nesnoj, 2017-08-29:
Good point, done in 41a09565b68bc71b771d8ca4fa7ba9b86af11264.
@abokam Please be aware that all data exported in the next run include closed rings! Please open switch disconnectors / circuit breakers manually if needed.

gplssm, 2017-09-01:
Thanks! Since 41a09565b68bc71b771d8ca4fa7ba9b86af11264 introduces grids with closed rings are returned by run_ding0(), we can remove ring closing from the examples, right?

Is anything more to do here? Otherwise we can close it and soon launch v0.1.3-:rocket:

nesnoj, 2017-09-01:
Yepp, could you please remove it?
I think there's nothing else to do.
I like your rocket :rocket: but there're some more tasks in #228 :smile_cat:

(please close as soon as you have adjusted the examples)

commit 00f2b240 (2017-09-01): Remove closing of CB's in examples

* This is now done in run_ding0()
* Finally fixes #224
closed 2017-09-01 by gplssm