openego/ding0 #225: RES: use new view instead of table for status quo scenario

nesnoj, 2017-08-25:
In DP v0.3 there will be only 1 table for RES and 3 mviews for sq, nep2035 and ego100 scenarios. In dingo, we must load the data from sq mview.
It'd be great to implement this prior to the next dingo run @gplssm.

nesnoj, 2017-09-01:
I prefer to implement it once dp v0.3 will have been released since the current MVIEWs are in an unstable state (`model_draft`). If we would implement it now, we would have to update our defs again..
@wolfbunke, @ulfmueller: Is there a current, definite deadline (not the past one in the [dp milestone](https://github.com/openego/data_processing/milestone/3)) for the dp v0.3 release?

If
- there's no definite deadline and
- we do not highly depend on new datasets in eDisGo  

I propose to postpone this issue to v0.1.4 and release v0.1.3 asap.

nesnoj, 2017-09-01:
I adapted dingo to use the recent mviews from model_draft, in branch [`features/adapt-to-powerplant-mviews-dp-v0.3.0`](https://github.com/openego/ding0/tree/features/adapt-to-powerplant-mviews-dp-v0.3.0).
Tested successful on 1 MVGD.
The dingo exports can now be used for https://github.com/openego/eDisGo/issues/21.

ulfmueller, 2017-09-05:
concerning the deadline for the dp, I hand over to @IlkaCu

IlkaCu, 2017-09-05:
I'll do my best to finish the dp release v0.3 before our project meeting next week.

commit 72a5c5c2 (2018-01-17): Use data processing version v0.3.0

* Include a change to mviews for structured data
* Fixes #225
closed 2018-01-17 by gplssm