openego/ding0 #226: Open/close circuit breakers BUG?

miguelparadacontzen, 2017-08-28:
If a new network is created, all its circuit breakers are open. If we use the grid method `rings_nodes()` then the circuit breakers are automatically closed and then a list of nodes in rings is returned.

The following function prints the status of the circuit breakers in a grid and print the list of nodes in its rings:
```
def test_circuit_breakers(nw):
    for district in nw.mv_grid_districts():
        for node in district.mv_grid._graph.nodes():
            if isinstance(node, CircuitBreakerDingo):
                print(node.__repr__() + ': ' + node.status)

        print('List of nodes in rings: ')
        print(list(district.mv_grid.rings_nodes(include_root_node=True)))
```

For some file 'dingo_tests_grids_1.pkl' with a network, the output of the following code
```
    nw = load_nd_from_pickle(filename='dingo_tests_grids_1.pkl')
    test_circuit_breakers(nw)
    print('####################')
    nw.control_circuit_breakers(mode='close')
    nw.control_circuit_breakers(mode='open')
    test_circuit_breakers(nw)
```
is:
```
circuit_breaker_2: open
circuit_breaker_3: open
circuit_breaker_4: open
circuit_breaker_1: open
circuit_breaker_5: open
List of nodes in rings: 
[]
circuit_breaker_1: open
List of nodes in rings: 
[]
####################
circuit_breaker_2: open
circuit_breaker_3: open
circuit_breaker_4: open
circuit_breaker_1: open
circuit_breaker_5: open
List of nodes in rings: 
[[lv_station_407595, mv_cable_dist_20, lv_station_408881, lv_station_62239, mv_cable_dist_49, mv_cable_dist_50, lv_station_62242, mv_cable_dist_48, lv_station_409588, lv_station_62084, mv_cable_dist_27, lv_station_62086, lv_station_62087, lv_station_410129, lv_station_410517, lv_station_410294, mv_cable_dist_8, mv_cable_dist_4, mv_cable_dist_24, mv_cable_dist_26, mv_cable_dist_7, lv_station_62080, lv_station_485813, mv_cable_dist_22, mv_cable_dist_23, lv_station_62078, mv_cable_dist_6, mv_cable_dist_52, mv_cable_dist_5, mv_cable_dist_18, mv_station_3545], [mv_cable_dist_19, lv_station_62177, lv_station_62186, mv_cable_dist_33, mv_cable_dist_32, lv_station_62224, lv_station_62225, mv_cable_dist_44, mv_cable_dist_43, lv_station_62228, lv_station_408994, lv_station_62226, mv_cable_dist_51, lv_station_408372, mv_cable_dist_21, lv_station_407660, mv_cable_dist_14, mv_cable_dist_46, lv_station_62237, lv_station_62233, mv_cable_dist_47, mv_cable_dist_45, mv_station_3545], [mv_cable_dist_9, mv_cable_dist_12, mv_cable_dist_15, lv_station_62089, lv_station_62103, lv_station_62099, lv_station_62098, mv_cable_dist_31, lv_station_62097, mv_cable_dist_29, lv_station_62094, lv_station_62093, mv_cable_dist_28, lv_station_62090, lv_station_62176, lv_station_407434, lv_station_62106, mv_cable_dist_11, mv_cable_dist_10, lv_station_407121, mv_station_3545], [lv_station_407432, lv_station_408860, lv_station_62219, lv_station_62220, mv_cable_dist_42, lv_station_62221, lv_station_62218, lv_station_62214, lv_station_62212, lv_station_62210, lv_station_62211, lv_station_62208, mv_cable_dist_40, mv_cable_dist_39, lv_station_62204, mv_cable_dist_38, mv_cable_dist_37, lv_station_62201, mv_cable_dist_36, lv_station_62197, mv_cable_dist_35, mv_cable_dist_17, mv_station_3545], [lv_station_407082, mv_cable_dist_16, lv_station_405899, mv_cable_dist_2, lv_station_405804, mv_cable_dist_1, mv_cable_dist_13, mv_cable_dist_34, mv_cable_dist_3, lv_station_62077, lv_station_405096, lv_station_405352, lv_station_405739, lv_station_405824, lv_station_62217, mv_cable_dist_41, lv_station_62216, mv_station_3545]]
circuit_breaker_1: open
List of nodes in rings: 
[[mv_cable_dist_2, lv_station_337207, mv_cable_dist_7, mv_cable_dist_4, mv_cable_dist_6, lv_station_410627, mv_cable_dist_3, mv_cable_dist_1, mv_station_3544]]
```

That is, if the function `nw.control_circuit_breakers(mode='close')` is no used, the `rings_nodes()` function does not work. Why?

########################
note that if in `rings_nodes()` we change line [158](https://github.com/openego/ding0/blob/dev/ding0/core/network/grids.py#L158) from `if circ_breaker.status is 'open':` to `if not circ_breaker.status is 'closed':` Then it works. Why?

nesnoj, 2017-08-29:
Hmm, this might be due to the equality operator `is` (cf. [stackoverflow](https://stackoverflow.com/questions/132988/is-there-a-difference-between-and-is-in-python)).
Please try if it works out with `if circ_breaker.status == 'open'` in line 158.

miguelparadacontzen, 2017-08-30:
Then it works...

nesnoj, 2017-08-30:
I replaced occurences in several files in a4971b5e0806a5558a3ab55c051f7725a82581b2

closed 2017-08-30 by nesnoj