openego/ding0 #230: package renaming

nesnoj, 2017-08-31:
Change package name to "egodingo". Ok, @gplssm? Other suggestions?

gplssm, 2017-09-01:
Nope :)

I suggest eDingo (available on pypi.python.org). Further I would like to merge the documentation branch prior to another renaming.

nesnoj, 2017-09-01:
Uhh, this must be senile dementia. Right, we favoured "eDingo".

gplssm, 2017-09-01:
Maybe pre-senile ;-)

boltbeard, 2019-01-15:
Is there some sort of naming scheme  or convention that must be upheld? 
I'd like to suggest "Network Topology Estimator" - "NeToEs" or
 "Distribution Network Topology Estimator" - "DiNeToEs" both are available on pypi.

nesnoj, 2019-01-16:
I'd have preferred a name that is similar to ding0, but as eGo is done by now and dependencies etc. are set, I'd avoid renaming. Therefore, I close this one..

closed 2019-01-16 by nesnoj