openego/ding0 #239: Calculate load/generation in subtree of node for all types of nodes

gplssm, 2017-10-10:
Currently, when peak load/generation at nodes of a MV ring is determined, values from branches of the ring are only considered [if branches start from cable distributors (CBs)](https://github.com/openego/ding0/blob/65bf0153d2889c0365157de1b65d046858a6c771/ding0/grid/mv_grid/tools.py#L86-#L104).

Consider branches starting from other node types as well.

nesnoj, 2017-10-24:
you did fix this one, right?

mltja, 2023-03-03:
This is okay because branches only starting from cable distributors.

closed 2023-03-03 by mltja

nesnoj, 2023-03-06:
Are you sure about that?
In the past we had valid layouts like this:
```
|
C----------------------------C--------------G--------------G--------------G
|                            |              |
|                            |              |
|                            G              G
| (Ring)
```
C - Cable distributor
G - Generator