openego/ding0 #240: Determine transformer parameters from v_sc/v_scr via defining custon PyPSA transformer type

gplssm, 2017-10-30:
Providing transformer parameters X,R absolute brings a lot of pain dealing with power flow analysis and comparing to other data. Hence, we can come around this issue by calculating transformer parameters directly from  v_sc/v_scr that is provided in literature.

### TODOs

1. Entirely replace csv load by loading primary trafo data, determine custom transformer types and changes the code to use these
2. Test implementation on the basis of eDisGo results for a few grid districts

nesnoj, 2017-10-31:
Could you please provide some more details on the pain brought by X and R?

gplssm, 2017-11-02:
R and X are well defined, see this comparison of a PyPSA/Pandapower 250 kVA standard transformer.

Values on the left side refer to a custom defined transformer, the values of the right hand side refer to the standard transformer provided by PyPSA.
```
name             250 kVA 10/0.4 kV (custom) 0.25 MVA 10/0.4 kV (PyPSA)
attribute                                                             
bus0                       trafo1_secondary           trafo1_secondary
bus1                         trafo1_primary             trafo1_primary
type                      250 kVA 10/0.4 kV         0.25 MVA 10/0.4 kV
model                                     t                          t
x                                 0.0378286                  0.0381576
r                                     0.013                      0.012
g                                    0.0024                     0.0024
b                                        -0                         -0
s_nom                                  0.25                       0.25
s_nom_extendable                      False                      False
s_nom_min                                 0                          0
s_nom_max                               inf                        inf
capital_cost                              0                          0
num_parallel                              1                          1
tap_ratio                                 1                          1
tap_side                                  0                          0
tap_position                              0                          0
phase_shift                             150                        150
v_ang_min                              -inf                       -inf
v_ang_max                               inf                        inf
sub_network                               0                          0
x_pu                               0.151317                   0.152632
r_pu                               0.051997                  0.0479969
g_pu                            0.000599995                0.000599996
b_pu                           -1.36181e-08               -1.37365e-08
s_nom_opt                                 0                          0

R in p.u. as calculated in eDisGo:  0.0945714015969 0.0953939201417
X  in p.u. as calculated in eDisGo:  0.0325 0.03
```
The last two lines show r_pu, x_pu as currently calculated in eDisGo by

r_pu = R / z_base = R / (V_primary^2/S_base)

with S_base being the nominal apparent power of the transformer.

See [the script](https://gist.github.com/gplssm/6663f053c96d716b3e9a14dec0f44eab) for more details.

Conclusion: we calculate r_pu, x_pu wrong, but not too badly. Anyway, should be fixed.... As long as we don't fully understand how PyPSA converts R to r_pu we would have to add the r_pu, x_pu attribute to the Transformer class or we override R, X with the pu values. In the latter case we have to think about how the grid's mean impedance can successfully calculated. Maybe it works as well in pu.

gplssm, 2017-11-23:
Parameter definition for transformer types 
[transformer_types_eDisGo.txt](https://github.com/openego/ding0/files/1499754/transformer_types_eDisGo.txt)

pfe and i0 is still missing...

AnyaHe, 2019-05-21:
It is recommendable to use the standard types! Note that x and r given in the standard literature in the example above as well as the ones used in Pypsa are also p.u. values! The table equipment-parameters_LV_transformers.csv therefore contains wrong values as they are not calculated in p.u. 
Precalculating values in csv-files instead of using standard values and processing them within the code is dangerous as it is hard to countercheck. I will open a new issue to approach this in order to only use literature values.

For the interested ones I will attach a csv file with the correct calculations of x and r.
[Erklärung Trafo.xlsx](https://github.com/openego/ding0/files/3203355/Erklarung.Trafo.xlsx)

mltja, 2023-03-03:
Seems to be solved in #296

closed 2023-03-03 by mltja