openego/ding0 #245: Comparison of identical Ding0 data does not always confirm equality

gplssm, 2018-01-13:
The [function for comparison of Ding0 networks](https://github.com/openego/ding0/blob/f46db597dfd4a9b193345880d4cda1d3764b7d23/ding0/tools/tests.py#L112-#L222) sometimes state branch IDs aren't the same.
This can be reproduced by comparing 5-20 datasets generated by exactly the same Ding0 code and input data. The result is [like this](https://github.com/openego/ding0/issues/242#issuecomment-357243175). The grid is most probably identical, but for some reason the IDs do not match.

**Hints**
* This issue is maybe simply related to different sorting while [ID setting](https://github.com/openego/ding0/blob/1bb0530cd4a28a89d2a95c2457aa0d615139b35e/ding0/core/network/grids.py#L280-#L299) in Ding0 and sorting in the [comparison function](https://github.com/openego/ding0/blob/f46db597dfd4a9b193345880d4cda1d3764b7d23/ding0/tools/tests.py#L112-#L222).
* Both, the branch setting method and [to_dataframe](https://github.com/openego/ding0/blob/f46db597dfd4a9b193345880d4cda1d3764b7d23/ding0/core/__init__.py#L1393-#L1503) use [graph_edges](https://github.com/openego/ding0/blob/857ed6bc031186eb839c11172171a0bc69fa426f/ding0/core/network/__init__.py#L203-#L234) that return a sorted list of edges (sorted by nodes).

**Tasks**
- [ ] Make sure Ding0 is deterministic regarding IDs of branches
- [ ] Revise [comparison function](https://github.com/openego/ding0/blob/f46db597dfd4a9b193345880d4cda1d3764b7d23/ding0/tools/tests.py#L112-#L222) to detect equality properly
- [ ] Maybe [comparison function](https://github.com/openego/ding0/blob/f46db597dfd4a9b193345880d4cda1d3764b7d23/ding0/tools/tests.py#L112-#L222) such that it always returns more information about inequality

nesnoj, 2018-01-15:
According to #97, this was not completely solved. Refer to https://github.com/openego/ding0/issues/40#issuecomment-268478299 for further details and proposed investigations.

mltja, 2023-03-03:
Already fixed

closed 2023-03-03 by mltja