openego/ding0 #253: Unexpected very long lines in MV grid

maltesc, 2018-03-13:
I just saw a wrong assignment of a generator or load to a MV grid very far off (see red line in 
[image].

This case is MV grid 792, Line_7920002

nesnoj, 2018-03-13:
Thanks for reporting!
Could you please provide the geno_id?
I tried to open your DB table `model_draft.corr_vis_mv_lines` to get more details but got no permissions..

maltesc, 2018-03-14:
The purpose of the `vis` tables simply is the visualization of my results and they constantly change. However, you can have a look at my tables `model_draft.corr_mv_lines_results` and `model_draft.corr_mv_bus_results` where I store all my eDisGo results. You should have access to those now. The mentioned line appeared in `result_id = 384`.
The connected generator is `Generator_433693`.
I hope this helps a little...

nesnoj, 2018-03-14:
Uhh, semms that there're some more lines affected:
[image]

The geno with `id = 433693` seems to be in Bavaria whereas HV/MV (`id = 792`) and MV/LV (`id = 285975`) stations are in Berlin (all data from dp v0.3.0). There seems to be an issue with the assignment of the geno to the right HV/MV station.

2 example genos:
```
SELECT *
  FROM supply.ego_dp_res_powerplant
  WHERE version='v0.3.0'
  AND id =in (433693, 19030);
```
Striking: Both genos are flagged (`rea_flag`) with M4_rest
@Ludee U got an idea?

mltja, 2023-03-03:
Check that an error is thrown if a generator with a too big distance is connected to the grid.

closed 2026-10-30 by nesnoj