openego/ding0 #255: Provide power_class to Fluctuating Generators

boltbeard, 2018-05-15:
From `model_draft.ego_renewable_feedin` the time series for `wind_onshore` has been divided into 7 power classes and a time series for each power class is created. This is assigned to every generator with every `wind_onshore` entry in `model_draft.ego_supply_pf_generator_single` being one from 1-7 and the rest of the technologies having a `power_class` of 0. More info about this on https://github.com/openego/data_processing/issues/231

This means along with solving issue #254 the `power_class` also has to be included in GeneratorFluctuating Objects to make sure the correct time series is assigned to them. 

Currently in the absence of the power_class, all `wind_onshore` are assigned a feedin time series corresponding to the `power_class` 4 until one is available

boltbeard, 2018-07-12:
### Tasks

- [ ] The materialized views need to include `power_class` of every generator
    - model_draft.ego_supply_res_powerplant_sq_mview
    - model_draft.ego_supply_res_powerplant_nep2035_mview
    - model_draft.ego_supply_res_powerplant_ego100_mview
    - supply.ego_dp_supply_res_powerplant_sq_mview
    - supply.ego_dp_supply_res_powerplant_nep2035_mview
    - supply.ego_dp_supply_res_powerplant_ego100_mview
@IlkaCu, @wolfbunke, @ulfmueller Could you please tell me how much effort would this take?

- [ ] A `power_class` attribute needs to be included in the GeneratorFluctuating Class
- [ ] Document the availability of this attribute

boltbeard, 2018-07-17:
Hi @wolfbunke, Before looking too deeply into the issue, I'm really interested in the answer to:
> 1. How are power classes dealt with in eTraGo? It would be good to handle them similarly in eDisGo for consistency.

Once I have some knowledge about this, it may be easier to see its criticality

The rest of the question are already to an extent answered by @MarlonSchlemminger in https://github.com/openego/eDisGo/issues/121

wolfbunke, 2018-07-17:
I see two option to map the power_class to the mViews.

1. Adding the column to the Mview by a join with table` model_draft.ego_renewable_feedin `
```
SELECT a.power_class, b.*
    FROM model_draft.ego_renewable_feedin a INNER JOIN (SELECT * FROM model_draft.ego_dp_supply_res_powerplant b WHERE b.scenario::text = 'Status Quo'::text 
	AND b.electrical_capacity > 0::numeric 
	AND b.preversion = 'v0.3.0'::text ) b ON (a.w_id = b.w_id) 
```

2.  Using `model_draft.ego_supply_aggr_weather` and adding power_class into the query
see here: https://github.com/openego/data_processing/blob/dev/dataprocessing/sql_snippets/ego_dp_powerflow_assignment_generator.sql#L416-L433 and using it. 

At the moment I was trying to run 1.) but I got few problems.

boltbeard, 2018-07-18:
Hi @wolfbunke, Based on the [sql code](https://github.com/openego/data_processing/blob/d0c2da7464e02af63edcab627dad934f757038c3/dataprocessing/sql_snippets/ego_dp_powerflow_timeseries_generator.sql#L179-L206) you have above, my estimate is that this task would take a higher priority.

> At the moment I was trying to run 1.) but I got few problems.

Thank you for making a start, let me know when you may likely be able to overcome the problems.

boltbeard, 2018-07-25:
An update to this issue has been discussed, the minutes are documented in [eDisGo issue #121](https://github.com/openego/eDisGo/issues/121), and its current state is summarized below:

> After having a closer look at the power_class implementations in eTraGo and its working within eGo in combination with eDisGo, the following conclusions were made:
> 
> 1. The major use case of eDisGo is to be used in conjunction with eTraGo as part of the entire eGo tool suite, where active power timeseries in eTraGo would be calculated by considering the power classes in the aggregated generators, and provide an averaged time series curve per technology per weather cell to eDisGo for each technology for each weather cell. (This is the current state of the implementation)
> 
> 2. The current state of the implementation does induce inacurracies due to the averaging of all the time series of all the generators at an mv grid. It is currently assumed that this induced inaccuracy is acceptable until quantifiable evidence proves it to be otherwise. Although this has not been quantified yet, there are methods that this can be done

Further information can be obtained by the issue directly

mltja, 2023-03-03:
Obsolete with new data.

closed 2023-03-03 by mltja