openego/ding0 #268: Aligned column names for CSV, OEDB, description and metadata

gplssm, 2018-07-19:
Column names for CSV/OEDB export must be changed according to [this proposal](https://docs.google.com/spreadsheets/d/1gSQ_H6zKo640TK705NeWBRRKsJkgKpcwdtk0g9AXX-w/edit#gid=1321404571)

It applies to multiple locations which must be done at the same time
  - [ding0 exporter](https://github.com/openego/ding0/blob/5f913f22d54bce1c3267558d23d4b7fe72eee2f3/ding0/tools/results.py#L2302-#L2513)
  - the [description of the tables](https://github.com/openego/ding0/blob/5f913f22d54bce1c3267558d23d4b7fe72eee2f3/doc/usage_details.rst) 
  - the [eDisGo grid import from CSV (OEDB)](https://github.com/openego/eDisGo/blob/119a69399af32b6a3c0b45c3b88af49c6fc9a244/edisgo/data/import_data.py#L2373-#L2690)
  - [ego.io ORM classes](https://github.com/openego/ego.io/blob/7ae69d3f555d586a57100d73e9eee55c3a7be165/egoio/db_tables/model_draft.py#L7361-#L7603)

jh-RLI, 2018-07-31:
tryed to reference this in my commit message, but couldnt make it work:
https://github.com/openego/eDisGo/blob/fix/ding0%23268-column-improvement/edisgo/data/import_data.py

mltja, 2023-03-03:
Closed due to age, reopen if needed.

closed 2023-03-03 by mltja