openego/ding0 #273: release v0.1.9

nesnoj, 2018-10-17:
We need a final eGo release.

ToDos:
- [ ] remove upper version bound of matplotlib dependency, test with recent version @nesnoj 
- [ ] fix example script `example_analyze_single_grid_district.py` @nesnoj 
- [ ] update docs: introduce oedialect @nesnoj 
- [ ] merge PRs #264 and #271 - i this really needed? @boltbeard 
- [ ] fix: LV voltage in kV in some grids @nesnoj 

Please amend if necessary.

@birgits Is it ok for you to have it done by Oct 29th?
https://github.com/openego/eGo/issues/141 states Oct 22nd/23rd but I'll be on vacation next week.

birgits, 2018-10-17:
@nesnoj Oct 29th is far too late as edisgo and ego depend on the ding0 release and releasing everything one day before the workshop seems too risky...

boltbeard, 2018-10-17:
>  merge PRs #264 and #271 - i this really needed? @boltbeard 
I would say #271 is stable and is very critical to have for any use-case of ding0 that assumes non-unitity power factor on anything (load or generation)

#264 is still not stable, it is still in the "Work in Progress" state, and I would not recommend merging it in its current state

nesnoj, 2018-10-17:
@birgits: Damn, I see my vacation to vanish.. :( I'll try to do this next week.
@boltbeard: I'll exclude those 2 PRs from the release

boltbeard, 2018-10-17:
@nesnoj, you mean 1 PR?

birgits, 2018-10-17:
@nesnoj how about you do point 1 of the list, @boltbeard does point 2 and merge of PR271, I do point 4 with help of @boltbeard and we drop points 3 and 5 for now? Point 1 should be fixed quite quick I guess. I would probably not use latest matplotlib version but the one suggested in networkx 1.11, which is 2.0.2.

nesnoj, 2018-10-21:
No. 1 is done

birgits, 2018-10-22:
Release is through! Thanks for your help!

closed 2018-10-22 by birgits