openego/ding0 #275: export features

nesnoj, 2018-10-17:
Thanks for the exporter, this makes life a lot easier.

While messing around with the export, I stumbled across 2 things:
(I used a network obj with several MVGDs)

1. Only 1 HV/MV station is exported. While you use indices for other components, you hard-coded this in stations:
https://github.com/openego/ding0/blob/c04e8eb07c5bc8d49d68a6a0a97c757640ee4619/ding0/io/export.py#L176-L185
2. The schema in db_export is hard coded: https://github.com/openego/ding0/blob/c04e8eb07c5bc8d49d68a6a0a97c757640ee4619/ding0/io/db_export.py#L41

Sorry for not creating a PR due to lack of time for testing..

nesnoj, 2018-10-19:
And another issue: while exporting some hundred grids, there was at least one which crashed during export of `lv_grids`. The origin is here:
https://github.com/openego/ding0/blob/ca6ca4e8ef86686430963dee460588f2769371de/ding0/io/export.py#L443
The geom is a Polygon but the column type is MultiPolygon. Without reviewing the code I have no clue why we have the wrong shapely geom type attached to `.geom` property.
I needed a fast fix so I just added
```
                        if isinstance(lv_district.geo_data, Polygon):
                            lv_district.geo_data = MultiPolygon([lv_district.geo_data])
```
prior to the dict assignment (MultiPolygon must be imported from shapely in the script header). This ugly fix must not be implemented in the branch, but it may helps you to avoid the export to crash for your run which I guess is quite time-critical too.

I can check why we have the wrong type here by end of Oct.

mltja, 2023-03-03:
Close because of upcoming major refactoring of the database interface #369, and then the using of saio.

closed 2023-03-03 by mltja