openego/ding0 #288: Voltage delta calculation at stubs while checking technical constrains

boltbeard, 2019-04-10:
There apears to be an issue with the application of consistent sign convention while calculating the  [`voltage_delta_vde`](https://github.com/openego/ding0/blob/dev/ding0/flexopt/check_tech_constraints.py#L529-L564) for stubs.

The function  [`voltage_delta_vde`](https://github.com/openego/ding0/blob/dev/ding0/flexopt/check_tech_constraints.py#L529-L564) expects a `+` or `-` sign in front of the `x` term (reactance term) as mentioned specifically [here](https://github.com/openego/ding0/blob/dev/ding0/flexopt/check_tech_constraints.py#L544-L547) 

In the functions that `voltage_delta_vde` is used i.e.:
 
1. To get the voltage change due to connections at [busbars](https://github.com/openego/ding0/blob/dev/ding0/flexopt/check_tech_constraints.py#L772-L781)
2. To get the voltage change due to connections at [branches](https://github.com/openego/ding0/blob/dev/ding0/flexopt/check_tech_constraints.py#L643-L646)
3. To get the voltage change due to connections on [stubs](https://github.com/openego/ding0/blob/dev/ding0/flexopt/check_tech_constraints.py#L716-L731)

In 1. and 2. this `x` term consistently gets a `-` sign when function is for generators (the assumption I believe is that they behave capacitively)

In 3. the `x` term doesn't get a `-` sign when applied for generators. It seems like an anomaly. Is there a specific reason why generators are treated differntly at stubs?

The current state is that for the issue #266 and PR #271, the sign change for the `x` term is dictated by the config input [`cos_phi_gen_mode`](https://github.com/openego/ding0/blob/fix/%23266-provide_powerfactor_mode/ding0/config/config_calc.cfg#L91) and [`cos_phi_load_mode`](https://github.com/openego/ding0/blob/fix/%23266-provide_powerfactor_mode/ding0/config/config_calc.cfg#L89). The PR assumes this anomaly as a mistake and fixes it. But to be sure of it, more information is needed. An answer is needed to be able to merge the PR #271 and close the issue #266.

gplssm, 2019-05-02:
I had a discussion with @boltbeard and we agreed that inconsistent use of signs, as explained above, does not make sense.

I confirmed that the _x_ that goes into [voltage_delta_vde](https://github.com/openego/ding0/blob/dev/ding0/flexopt/check_tech_constraints.py#L529-L564) is always positive in all three cases (sub-functions that are mentioned above). Based on this @boltbeard says, the (-) sign should be used consistently.

gplssm, 2019-07-08:
Closed by ac23f710a49ca74352761ec82f2385c0d464984f. See commit message

closed 2019-07-08 by gplssm