openego/ding0 #292: Use geodesic distance calculations instead of vincenty from geopy

boltbeard, 2019-05-07:
The geographic distances in the functions are calculated using the vincenty algorithm in the pacakge geopy:
- [calc_geo_dist_vincenty](https://github.com/openego/ding0/blob/dev/ding0/tools/geo.py#L119-L152)
- [calc_geo_dist_matrix_vincenty](https://github.com/openego/ding0/blob/dev/ding0/tools/geo.py#L155-L203)

Geopy in their [documentation](https://geopy.readthedocs.io/en/stable/#module-geopy.distance) have in the recent past changed its default distance calculation to geodesic as it is better (converges always) 

It would be prudent for us to use the same methods. It may also be good to use distance.distance instead of distance.vincenty

nesnoj, 2019-05-07:
Concerning the accuracy (high enough) or non-convergence (the grid extends will hardly exceed 20000 km ;) I do not see a reason to change the method. But as vincenty [will be removed](https://geopy.readthedocs.io/en/stable/changelog_1xx.html#id7) in geopy v2.0, I agree!
Go for it

nesnoj, 2021-05-19:
Done as part of #325

closed 2021-05-19 by nesnoj