openego/ding0 #306: Passwordless input data query

gplssm, 2019-09-02:
Since version [0.4.7](https://github.com/openego/ego.io/releases/tag/v0.4.7), ego.io support passwordless data query from OEP. Let's use this feature in Ding0 to make the usage of Ding0 easier to understand.

* [ ] Update ego.io version in dependency list
* [ ] Include "Close Session" statement (-> see implementation in eDisGo)
* [ ] Replace all session usages in entire code base (in particular examples)
* [ ] Replace usage of Ding0_test_user to query data for tests
* [ ] Update docs:
  - Describe PW-less as default
  - Describe briefly how to use a config file

Implementation ideas

- put session creation in ding0.tools.__init__.py
- handle query error with context manager (see eDisGo)
- import session from ding0.tools whereever neeeded

birgits, 2019-09-17:
In 055ff41538af826025904b54914b6b2cc83db1e4 I updated the egoio version and changed all connections to readonly. I didn't include close session etc. yet. 
Further, in 2b9d7b4cf742f87dc888b07b3a2d0b3024218faf I deleted the section about the oedb config in the documentation as I think it's not necessary anymore. Maybe we can include it again in the developer section as it might be necessary when grids are exported to the oedb?

nesnoj, 2021-05-28:
Seems that this is not longer relevant, right @gplssm @birgits ? In case that we wanna write results to the OEP, we can open a new issue.

closed 2021-05-28 by nesnoj