openego/ding0 #320: Grid export to csv fails

AnyaHe, 2019-12-12:
When using the new version of #312 , some of the grids cannot be exported as csv-files.

AnyaHe, 2019-12-13:
Empty LV grids are ignored now.
The following grid districts still cause an exception:
473, 931, 1822, 1879, 1944, 2714.
Even though they do not contain any loads or generators, the assigned peak_load or peak_generation is not equal to 0. A check why this is the case, should be conducted. As a solution for the export, the exception can be adjusted in a way that it is checked whether the LV grid contains any loads or generators or not.

nesnoj, 2021-05-28:
I checked the data and it seems that affected MVGDs contain at least one load area without an LVGD. Example MVGD 437:

[image]

This was already detected in #155 and origins from https://github.com/openego/data_processing/issues/68.
Code links:
- ding0 fails prior to export here: https://github.com/openego/ding0/blob/2b8021751e41c1f6a3e51141ac34540cf7a00a8b/ding0/grid/lv_grid/build_grid.py#L791-L797
- Fallback snippet for LA without LVGD: https://github.com/openego/ding0/blob/2b8021751e41c1f6a3e51141ac34540cf7a00a8b/ding0/core/__init__.py#L496-L526

As #346 will introduce new MV-LV substations and LV grid districts, the ones from OEP will get obsolete. I'll leave the issue open though.

mltja, 2023-03-03:
Fixed or obsolete with new data and actual new_dev branch.

closed 2023-03-03 by mltja