openego/ding0 #325: Python 3.7 and 3.8 are not supported

gplssm, 2020-01-29:
Adding support requires

- [ ] Installation works properly
- [ ] Tests passing 
- [ ] SPDX classifier update
- [ ] travis file includes new python versions

nesnoj, 2021-05-18:
@gplssm With py3.8 and the [recent packages](https://github.com/openego/ding0/blob/a4294bae262058ee77fde664eb63ab3c53ec1f36/requirements.txt) it seems to work fine (some tests are still running, though).

The only packages with version limits are `ego.io` (explicit commit) and `sqlalachemy` (due to oedialect incompatibility). Shall we set versions for the other packages too? For at least some packages I'd not expect newer versions since py3.8 is not the latest python version. But it may help to avoid problems in the future.

nesnoj, 2021-05-19:
As the tests seem to be sensitive to package versions, I'll restrict them.

PR #340 (2021-05-28): Features/#325 add python3.8 support
Fix #325
closed 2021-05-28 by nesnoj