openego/ding0 #326: sum capacity of grid generators

azizmassano, 2020-04-22:
I've problems running ding0 examples 'example_single_grid_district.py'
when I run run_ding0 it results: 'TypeError: unsupported operand type(s) for +: 'int' and 'str'
specifically, the problem occurs when:
    cum_mv_peak_generation = sum([_.capacity for _ in self.grid.generators()])
capacities of generators are string: for example '580', and they cannot be summed together.

can somebody knows if there is an internal error or if I'm doing something wrong?

gplssm, 2020-06-08:
Sorry for reacting that late! Thanks for reporting!

Seems to me like it's due data being delivered differently. I'll try to fix it.

gplssm, 2020-06-09:
This was quite a nasty thing. Numeric data retrieved from the database was casted into a string. Thus, you ended up with this error, because the sum over an array of strings can't be computed.

The fix will close this error. Then, installing the dev version should run again.

PR #331 (2020-06-09): Use updated (unreleased) version of ego.io with fixed type
Fix #326
closed 2020-06-09 by gplssm