openego/ding0 #332: Missing bus / lv grid

birgits, 2020-07-27:
I wanted to check out the current ding0 grids generated with the refactoring branch, wherefore I generated all grids for the whole of Germany and then tried to conduct a worst-case analysis using edisgo. For alot of grids I get the same error upon importing the ding0 grid with edisgo, e.g. for grid 1: "The following transformers have bus1 which are not defined: Transformer_lv_grid_506255_1.". The error pops up when it is checked whether each component's specified bus(es) exist in the buses dataframe. In the case of grid 1 there is a transformer for which only the bus on the mv side exists but not the one on the lv side. This is probably due to no lv grid being generated in ding0. I am wondering why that is? It could be a problem on the data side or of the export (because upon creating the ding0 grid, ding0 gives the message that there are no voltage issues wherefore I assume that a power flow analysis can be conducted, wherefore in ding0 it seems to be a working lv grid). My next question is how we should cope with this issue? I could temporarily fix it in edisgo in case it is a greater issue that needs to be fixed in ding0.

AnyaHe, 2020-08-10:
I think, the problem is, that there are some empty grids (raising the info in build_grid.py line 800). As no grid is build around them, the station of LV side is an islanded node. In order to be able to export the grids anyways, I introduced the workaround in pypsa_io.py line 328. When exporting only MV, an artificial node is created at the end of each transformer, which is probably why running power flows in ding0 still worked for these grids. However, when exporting mv and lv grids, the LV side of the transformer is not exported as isolated nodes are ignored. I extended the workaround in a way that introduces the LV side of the transformer to the node_df if lv grids are exported as well. However, this does not fix the problem why there seem to be empty grids. In a later version, this should be overthought.

AnyaHe, 2020-08-10:
Is fixed in #312

commit 2c415db6 (2020-08-10): fix #332: if lv grids are exported as well, export secondary side bus for isolated lv stations
closed 2020-08-10 by AnyaHe