openego/ding0 #334: Installation with .yml fails and errors when running examples first time

IsGut, 2020-11-30:
After installing and first run on windows:
In core\networks\stations.py, line 58: cum_mv_peak_generation = sum(...):
Type Error: unsuported operand type(s) for+: 'int' and 'str'
=>_.capacity is str and need to be casted, e.g. in core/__init__.py

In Abschlussworkshop open_eGo 30. Okt. 2018 Session:
At 5.6 Komplexe Lastflussberechnung durchführen plot_mv_topology(...) error occurs:
'You must first set_array for mappable'. After setting: np.array([]); error occurs:
UserWarning: no labelled objects found. Use label='...' kwarg on individual plots. warnings.warn("No labelled objects found. ")

After running first example from readthedocs a .pkl file is generated how it is explained. While running analysis script I get errors when trying export key figures:
TypeError: 'str' object cannot be interpreted as an integer (maybe due to my casting of .capacity?)
KeyError: 'L_l'

Installation on fresh Ubuntu OS with ding0_env.yml doesn't work:
Error: Failed building wheel for pyproj
Error: Failed building wheel for pandas
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1 

Because ding0 is working for my goals and these mentioned errors doesn't affect me for now, I can ignore them.

serpoten, 2020-12-03:
Interesting, I have have if not the same error, then close. 

If I follow the steps from the documentation it actually does not work the environment purposed. 

I managed to get it install in a Debian machine, with an anaconda environment using a specific python version, 3.6.5 (according to the .yml file from GitHub).

Once I installed, I tried running the example, " example_single_grid_district.py" it does not work.

I get  the following messages,

- **Warning**

`SAWarning: Ignoring declarative-like tuple value of attribute segment: possibly a copy-and-paste error with a comma left at the end of the line?
  "left at the end of the line?" % k)`

- **Error**

`TypeError: unsupported operand type(s) for +: 'int' and 'str'`

Complete file log: [log_frist_run.txt](https://github.com/openego/ding0/files/5638664/log_frist_run.txt)

nesnoj, 2020-12-04:
Hey @IsGut and @serpoten, thanks for your hints! We'll try to figure this out and get back to you..

serpoten, 2020-12-04:
Hi @nesnoj, I just tried a different approach and it works.

1. Create an anaconda environment with specific version 3.6.5

2. Clone the repository into your computer

3. Activate the environment and navigate to the folder where you have the clone repository `cd path file`

4.  Instead of installing ding0 through `pip install ding0` use `pip install . ` since there is a setup file.

5. Navigate to the example folder in your cloned file `cd path file`

6. Run the examples in order. You should get the ".pkl" file and which is necessary for the analsysis file.

So this makes me believe that the pip version installs an outdated version with some minor bugs, since installing it from the git information it actually runs smoothly.

gplssm, 2020-12-08:
To both of you, thanks for reporting these issues! Core of all the reported issues are the versions of depending packages and the numerous upper version limits, i.e. `pandas<=0.23.0`.

I could verify the successful installation following @serpoten 's approach. Steps

```
conda create -n ding0_env_test_2020-12-08 python=3.6
conda activate ding0_env_test_2020-12-08
pip install git+https://github.com/openego/ding0.git@dev # does clone + install
```

Please pay attention that I've specifically specified Python 3.6. I tried with Python 3.7 and it didn't work because (I assume) `pandas<=0.23.0` isn't available anymore in Python 3.7.

The reason why @serpoten had success installing ding0 with cloning and installing with `pip install . ` but not with `pip install ding0` is that different versions are installed. `pip install ding0` looks in the index of pypi.org and installs the latest **published** version. With cloning and installing with `pip install . ` the code from the dev branch (default branch) is installed.
In order to make `pip install ding0` work again, a new version needs to be released on PyPI.
This is planned (since long time), but first a long time pending feature branch needs to merged and the installation issues need to be fixed entirely. I gonna do this, but I assume I won't make it this year.

IsGut, 2020-12-09:
You mention `pandas<=0.23.0` but in .yml ist written 
```
# numerical processing packages
- pandas = 0.20.3
```
So, can I update to `pandas=0.23.0`?

nesnoj, 2021-05-28:
The installation should work out of the box now with [ding0 v0.2.1](https://github.com/openego/ding0/releases) (also available on [PyPI](https://pypi.org/project/ding0/)), I tested on xUbuntu 20.04 and Windows 10 via conda #339.
Please reopen if you encounter problems.

closed 2021-05-28 by nesnoj