openego/ding0 #358: Run ding0 on custom area

azizmassano, 2021-09-09:
Hi. I'm new to ding0. I've managed to run the example files.
I'm trying to apply the tool to another site, different from the region in the examples.
Can somebody show me an example of the application of ding0 in another location?
Do I must create an OEP db or can I insert the shapefiles of my region 'by hands'?
Thanks,
Marco

nesnoj, 2021-09-10:
Hello @azizmassano, ding0 is based upon data created with the open_eGo [data_processing](https://github.com/openego/data_processing) which defines so called MV grid districts (MVGD) (cf. paper [Huelk2017] in [docs](https://dingo.readthedocs.io/en/dev/about.html)) with specific data on those regions. It is currently not possible to use custom regions.
But you can use one or more MVGDs which cover your region of interest. The dataset on MVGDs can be obtained manually from the OEP from [this table](https://openenergy-platform.org/dataedit/view/grid/ego_dp_mv_griddistrict) in different formats and plotted e.g. with geopandas or QGIS (there're some [tutorials](https://openenergy-platform.org/tutorials/) on how to do this) to find out the MVGDs of interest. The dataset contains different versions, make sure you filter for the latest version (v0.4.5).

Hope that helps

PS: It's not necessary to set up an "OEP db", ding0 obtains the data from the OEP.
PPS: We're currently working on a better data basis, urban MV grids and routed LV grids in [eGon-data](https://github.com/openego/eGon-data/) (in the [eGo^n](https://ego-n.org/) project).

azizmassano, 2021-09-10:
thanks @nesnoj.
So it is not possible to extend the use of ding0 tool to one region not yet covered by the e_g0 project, right?
There is no way to replicate the simulation for instance in Italy, even if are known the input data sources needed by the tool (e.g., region boundaries, load areas, MV-LV substations)?
"But you can use one or more MVGDs which cover your region of interest". So I can run ding0 only if my region of interest is covered by the OEP dataset..
Please, correct me if I'm wrong and if I still don't understand.
Thanks, macro

nesnoj, 2021-09-10:
> So it is not possible to extend the use of ding0 tool to one region not yet covered by the e_g0 project, right?

At least not straightforward, see below

> There is no way to replicate the simulation for instance in Italy, even if are known the input data sources needed by the tool (e.g., region boundaries, load areas, MV-LV substations)?

Well, it's not impossible. But you
- either would have to run the data_processing (see above) for Italy. But as it uses quite some Germany-specific data that'd be some work and I don't know if that data would be available for Italy.
- or you create the [required tables](https://github.com/openego/ding0/blob/5ace85657f7fb1da45524321e7e1e0d65d3a4bce/ding0/config/config_db_tables.cfg#L19-L27) manually in the very same format as found in the [DB table definitions](https://github.com/openego/ego.io/tree/dev/egoio/db_tables) (you can find the data in the respective [schemas on the OEP](https://openenergy-platform.org/dataedit/schemas)).

PS: by the way, there has been some data on distribution grids for Italy some years ago in [Atlantide](https://www.progettoatlantide.it/Atlantide/jsp/index.jsp) but the link seems to be dead now..

azizmassano, 2021-09-10:
ok, thanks
I would ask you, if it is not a problem, your e-mail, to contact you for further details. Just to understand if it is a viable option or if I have to change the tool. 
Maybe we could have a skype-call?
I hope I'm not disturbing you,
Marco

nesnoj, 2021-09-13:
> I would ask you, if it is not a problem, your e-mail, to contact you for further details. Just to understand if it is a viable option or if I have to change the tool.
> Maybe we could have a skype-call?

I'm afraid that I cannot offer a Skype call but you can send me an email if you like: jonathan.amme@rl-institut.de

closed 2021-09-14 by azizmassano

LorenzoBottaccioli, 2023-03-08:
@nesnoj where can I find a documentation on how to run [data_processing ](https://github.com/openego/data_processing)for a particular country?