openego/ding0 #359: Unnecessary extra buses and lines in lv grids

khelfen, 2021-09-23:
Residential loads and generators are now connected to the same bus within lv grids, but there seems to be extra buses and lines downstream from loads without any function.

See this example lv grid with all loads marked bigger:
[image]

and this example with all generators marked bigger:
[image]

Downstream of each load there is another bus with nothing connected but the line from the load. See this example:
[image]

nesnoj, 2021-09-24:
Thanks for reporting, I think @AnyaHe modified the connection of residential loads and feeders, maybe she can help?

Is it crucial for your current work? I'm asking since the LV model grids will be replaced anyway by road-routed grids in the next months.

khelfen, 2021-09-24:
It's not crucial :)

mltja, 2023-03-03:
Solved in new_dev, reopen if still exists with new grids.

closed 2023-03-03 by mltja