openego/ding0 #362: floating point error in local search operator oropt

piaulous, 2022-01-19:
In some rare cases the `operator_oropt` function had not come to an end, because it was stuck in the while loop.

I solved it myself by adding

`length = round(length,5)`
` length_best = round(length_best,5)`

after line 203.

https://github.com/openego/ding0/blob/5ace85657f7fb1da45524321e7e1e0d65d3a4bce/ding0/grid/mv_grid/solvers/local_search.py#L182-L211

nesnoj, 2022-01-20:
Thanks for reporting @piaulous!
Do you encounter this every run?

So the break condition in L211 is never fulfilled. Hence there's always a better solution found as the only place where `length_best` is modified is in L201. So the tour should be modified in L205 but something goes wrong here, I propose to start debugging right there.

piaulous, 2022-01-21:
Yes, it occurs for every run and for the same mv grid districts.

I'm quite sure, that the optimal solution already has been found, but the number of decimal places of `length` and `length_best` sometimes differ, so that the break condition at L211 could not be fulfilled. That's the impression I had when I did the debugging.

mltja, 2023-03-03:
fixed and in new_dev

commit 71543ed0 (2022-03-30): fixes #362