openego/ding0 #391: Meshed LV grids

piaulous, 2023-05-08:
A very rare case of a meshed lv grid has been detected (one grid at the moment).
TODO's
1. check origin of this problem
2. if necessary right a function to get rid of the mesh

joda9, 2026-09-23:
Measured the scope on a full ding0 export — **3648 MV grids, 352768 LV grids**.

**159 LV grids are meshed, in 153 MV grids.** So it is rare per LV grid (0.045 %)
but hits **4.2 % of MV grids**, and a calculation always processes a whole MV grid.

21 of those 159 are not real meshes but a line whose two ends are the same bus — a
separate artifact, filed as #406. The remaining **138 are genuine rings**.

An example, `LVGrid_8829100000` in `mvgd_32064`: 41 buses on 41 lines, one triangle of
three cable cabinets.

| line | length | type |
|---|---|---|
| KVS 3 – KVS 4 | 374.5 m | NAYY 4x1x50 |
| KVS 3 – KVS 5 | 152.6 m | NAYY 4x1x50 |
| KVS 4 – KVS 5 | 222.0 m | NAYY 4x1x50 |

No switch, no open point, nothing marking it as a ring meant to be operated open.
Worth contrasting with MV, where ding0 is consistent: that grid's MV level has 120
buses on 119 lines and no cycle at all, because the rings are modelled open via the
circuit breakers.

**It does cause real breakage downstream.** eDisGo's pseudo-coordinate layout assumes
a tree; all 138 genuine rings made it abort with
`NetworkXError: The node ... is not in the graph`, taking the whole spatial complexity
reduction with them — **132 MV grids, roughly one in 28**. Fixed on the eDisGo side in
openego/eDisGo#765, but the grids still carry the rings.

joda9, 2026-09-23:
These are the **138 genuine rings** — the 21 self-loop lines of #406 are excluded, so this is the set that actually makes a tree-based traversal fail.

<details><summary>138 meshed LV grids in 132 MV grids</summary>

```
Genuine meshed LV grids in the ding0 export
138 LV grids in 132 MV grids
(self-loop artifacts of openego/ding0#406 excluded)

mv_grid   lv_grid           buses  lines  rings
30945     2186900000           43     43      1
30946     6770000000           52     52      1
30961     3272500009          132    132      1
30990     2127040000           71     71      1
30991     9988600032           74     74      1
31050     1560650004          214    215      2
31099     6839000004           56     56      1
31170     1458110001          175    175      1
31267     6245000099          212    212      1
31291     6281700067           42     45      4
31326     1193300000          175    175      1
31344     5963500002           69     69      1
31345     5978100011          144    144      1
31412     2096950006          193    193      1
31466     8935000000           27     27      1
31470     6468900000           55     55      1
31524     1695270001          329    329      1
31575     1902110000          316    316      1
31576     1987680005           58     58      1
31577     1510200000          200    200      1
31585     7718000000          146    147      2
31590     2359100001           21     21      1
31590     2359400000           35     36      2
31592     1532300000           56     56      1
31592     1612700000          114    115      2
31593     1650900002           47     47      1
31655     6284400000           98     98      1
31790     1000850045          114    114      1
31791     5691300014          141    141      1
31792     4556700013           63     63      1
31809     6967900000           27     27      1
31864     3897000000           17     17      1
31867     3289000000           18     18      1
31868     1289100001          101    101      1
31943     1325040001          222    222      1
31972     4272300000           31     31      1
32004     2110800007          373    373      1
32036     4324400000          106    106      1
32064     8829100000           41     41      1
32238     3231300000           51     52      2
32247     8216000000          107    107      1
32261     8021900002          151    151      1
32288     1259760000           28     28      1
32362     6948400093          162    162      1
32396     1173560000           43     43      1
32452     1293200001          193    193      1
32453     1697900000           48     48      1
32501     9164700000           19     19      1
32508     4185300000          177    177      1
32542     1917460078           38     38      1
32545     1169950000           50     50      1
32591     1949160004           59     59      1
32636     1635330000           83     83      1
32744     2658100000          279    279      1
32744     2946400054           33     33      1
32746     7577000041          181    181      1
32794     1869040002           29     29      1
32806     1542700002          220    220      1
32862     1132080001           76     77      2
32975     9628200000          106    106      1
[... 56 lines cut ...]
34632     6846200000           14     14      1
34644     7832800005          205    205      1
34735     3178400050           89     89      1
34786     1879650000          126    126      1
34812     1015490010           97     97      1
34873     9339200000           40     40      1
34917     3502200005           32     32      1
35133     3561000033           18     18      1
35134     7171700000          107    107      1
35152     3040700013           38     38      1
35157     1367800000          101    101      1
35165     2889400000          240    240      1
35168     1947900002          142    142      1
35208     1332620012           87     87      1
35659     1076000006           59     59      1
35670     8315000000          168    168      1
35733     3087400000          101    101      1
35801     1777170003          213    214      2
36067     2079760000          171    171      1
36121     2184490002          667    667      1
36137     4639600000          211    211      1
36165     1890860000           82     82      1
```

</details>

closed 2026-10-30 by nesnoj