openego/ding0 #400: Ding0 does not run with oedialect anymore with recent versions

maurerle, 2024-02-13:
The `NetworkDing0` now needs a engine as param - which is not yet changed in the tests.

The oedialect does not seem to work anymore with latest versions (see https://github.com/OpenEnergyPlatform/oedialect/issues/47).

I tried to fix the packages using newer versions, but ran into the issue that NetworkDing0 did not have a engine at first,
it then did not like oedialect and if I catch this error I run into:

```
Traceback (most recent call last):
  File "/home/admin/ding0webservice/main.py", line 288, in <module>
    service = Ding0Webservice()
              ^^^^^^^^^^^^^^^^^
  File "/home/admin/ding0webservice/main.py", line 46, in __init__
    self.ding0_builder = Ding0Builder()
                         ^^^^^^^^^^^^^^
  File "/home/admin/ding0webservice/utils/ding0_builder.py", line 37, in __init__
    self._ding0 = NetworkDing0(self._session, name='network')
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/ding0/ding0/core/__init__.py", line 167, in __init__
    self._orm = self.import_orm(session)
                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/ding0/ding0/core/__init__.py", line 1505, in import_orm
    engine = database.get_engine()
             ^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/ding0/ding0/tools/database.py", line 45, in get_engine
    f"{int(cfg_ding0.get('database_credentials', 'port'))}/{cfg_ding0.get('database_credentials', 'name')}",
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'port'
```

I therefore could not get my application to run with new versions.
Could you help with that?

jh-RLI, 2024-04-12:
Hey, sorry for the late reply. At the moment this repo is a little out of our current focus. This is mainly because we had to stick to Python 3.6 until recently (the OEP website was running on this version), which prevented us from further updating the sqlalchmey-related code. This problem has finally been solved this week.

I will have a look at the issues here.

nesnoj, 2024-04-15:
> Updating the sql-achemy version in oedialect has not been possible until now. We can take a look at it and maybe update it. My resources to work on it are rather limited at the moment. I will do my best ;)

Thank you @jh-RLI !

Leonrtg, 2025-09-09:
Hi @all 

I'm currently writing a Master Thesis about Open-source tools for georeferenced modelling of synthetic power distribution network topologies and I wanted to use ding0 as the basis for further investigations. 
I think I have the same issue with latest ding0 version on python 3.10. 

Can someone recommend me a workaround to this issue? 

I would be very thankful for every help. 

Best, Leon

jh-RLI, 2025-09-09:
Hi Leon,

in general, the oedialect still works. Last time i installed ding0 was years ago. Currently, i can't really put time into it to find the root of this error :/

One good thing is that there is work happening to update the SQLAlachemy version. But this might take weeks.

jh-RLI, 2025-09-17:
Hi  @Leonrtg sorry that SSL issue regarding the certificate chain was a bug which i fixed yesterday ... so this was some extra struggle you normally should not worry about. This means you should not need this anymore `VERIFY = os.path.expanduser("~/oep_ca_bundle.pem")` etc.

ChatGPT seems to overthink here :) You dont need to first open the connection and then access the get_columns API-endpoint.

You should be able to just send a request there like:

```python
import requests

PRODCTION_OEP_URL = "https://openenergyplatform.org"
REST_API = "/api/v0"
ADVANCED_API_HAS_TABLE = "/api/v0/advanced/has_table"

query = {"table": "all_ghg_emissions_by_sector", "schema": "model_draft"}

response = requests.post(
    url=f"{PRODCTION_OEP_URL}{ADVANCED_API_HAS_TABLE}",
    json={"query": query},
)
```

It is important to send as json data and package everything under this "query" keyword. This would usually be described in a good documentation, which is incomplete currently ....

 json={"query": query}, 

The `/advanced/connection/open` endpoint is implemented for the oedialect to start connections you could use it like that but it is currently not intended to be used like you did.

To answer your questions:

1. see above :)
2. Yes. Not all API endpoints need authentication. But if they need you add a header to your requests and you have to register on the openenergyplatform.org, navigate to your profile where you find your api token. 

```python
# Setup the header, replace ***** with your OEP API token
PRODCTION_HEADER = {"Authorization": "Token ****"}
...

response = requests.post(
    url=f"{PRODCTION_OEP_URL}{ADVANCED_API_HAS_TABLE}",
    headers=PRODCTION_HEADER, # - add the header to the request
    json={"query": query},
)

```

3. The relevant versions here are python==3.10 & sqlalchemy==1.3 the other package use their latest version and they are limited by the python and sqlalchemy version we use in the OEP ... ding0 only works on python 3.8 with the oedialect (i think). In most cases you can still use more recent python versions but especially when working with sqlalchemy you should stick to <= 3.10

4. No there is no fallback, the advanced endpoints are required to make the sqlalchemy dialect work with the OEP. But you can get the data without using a ORM as JSON data form the API.

Leonrtg, 2025-09-17:
Hi @jh-RLI, 

wow thank you so much for your quick reply! I think I came closer to make Ding0 work on my computer! 
Again, as I'm not an expert, I let ChatGPT summarize the so far progress I could make and the problems I occurred. 

Thank you again for your availability and help! 

ChatGPT Output: 

1) What works
- TLS/HTTPS: Trust is fixed; requests are correctly sent via https://openenergyplatform.org.
- Basic endpoints (Schema/Rows):  
  GET /api/v0/schema/grid/tables/ego_dp_mv_griddistrict/rows?limit=1 → 200 OK (after TLS fix).
- Advanced metadata (partially):  
  advanced/get_columns and advanced/get_pk_constraint return 200 OK when run without Cursor (see logs).

2) What does NOT work (blocker)
- advanced/cursor/open → 401 "Key 'connection_id' not found in {user: <...>}".
- In ding0 init (SQLAlchemy reflection via oedialect) it crashes as soon as the dialect needs a cursor (OECursor):  
  → error: Could not open cursor: Key 'connection_id' not found ...
- Browser-like login via POST /accounts/login/ (username/password) fails in scripts:
  - Login-POST: 404
  - "profile probe" on /user/profile/ → 404

3) New findings / cause hypotheses
- Payload form for Advanced API must (still) be sent as JSON with a "query" wrapper:  
  json = {"query": {...}} (confirmed by OEP side).
- Not all advanced endpoints require auth; some do. Auth via header: Authorization: Token <OEP_TOKEN>.
- The cursor workflow of the oedialect advanced API seems session-/login-bound:
  - connection/open does return an ID (200),
  - cursor/open, however, denies (401), apparently because the server session has no valid logged-in user assigned.
- The script login used "next" and probe path /user/profile/ (404). On OEP the profile is at /accounts/profile/.  
  → Wrong target path explains the 404 after login and why no logged-in session is created.

4) Where it specifically fails
- Ding0 needs the oedialect cursor path for reflection; this apparently requires a real, cookie-based login session
  (or a server-side session binding), which currently does not succeed.
- Token header alone does not seem sufficient for cursor/open (even though Authorization is sent in session headers).

5) Potential problems (API/client mismatch)
- OEP advanced contract: some handlers (especially cursor) may expect a logged-in session + "query" envelope;  
  oedialect uses these endpoints strictly and aborts on 401.
- Login flow in script: POST /accounts/login/ was correct, but "next" and probe URL wrong (/user/profile/ instead of /accounts/profile/).
- If your OEP account uses SSO (RegApp/OIDC), username/password POST technically fails (like in UI: "Sign in through your institution");  
  then a pure requests-based OIDC flow is much more complex.

7) Short conclusion
- Basic API works; advanced metadata partly works; cursor/transaction endpoints block without a valid login session.
- Main lever: establish a correct cookie-login (no OIDC) and/or confirm whether cursor endpoints accept token auth.

jh-RLI, 2025-09-18:
This Sounds a bit like you have issues to provide your OEP Token to ding0? Im Not on my PC today. Maybe you can Share how you run ding0?

jh-RLI, 2025-10-16:
not sure if this is correct

```
# python store_oep_token.py
import keyring
SERVICE = "oep"      # conventionally used by egoio
USERNAME = "xyz"
TOKEN = "xyz"
keyring.set_password(SERVICE, USERNAME, TOKEN)
print("Token stored in keyring.")
```

I think you should have a config file in your user directory. In terminal try to use `cd ~` and then look for a directory called `.egoio` in there should be a config file `config.ini` with the following:

```
[oep]
dialect = oedialect
username = Jonas.Huber
host = openenergyplatform.org
port = 80
database = oedb
password = ***********
```

jh-RLI, 2025-10-22:
Did you setup the logger already? Can you see if you send data to `openenergyplatfrom.org` or `openenergy-platform.org`? Im pretty sure you are still sending the query to the wrong host as i can reproduce your error when i change it to the version which includes the "-". Maybe the ding0 config file you mentioned collides with the config in `.egoio`.

Here is my full script:

```python
#!/usr/bin/env python3

# ===== IMPORTS AND CONFIGURATION =====

# import DB interface
import os
from egoio.tools import db

# import required modules of DING0
from ding0.core import NetworkDing0
from ding0.tools.logger import setup_logger
from ding0.tools.results import save_nd_to_pickle
from sqlalchemy.orm import sessionmaker
import oedialect

os.environ["OEDIALECT_PROTOCOL"] = "https"

# define logger
logger = setup_logger()

# ===== MAIN =====

# database connection/ session
engine = db.connection(readonly=False)
session = sessionmaker(bind=engine)()

# instantiate new ding0 network object
nd = NetworkDing0(name='network')

# choose MV Grid Districts to import
mv_grid_districts = [460]

# run DING0 on selected MV Grid District
nd.run_ding0(mv_grid_districts_no=mv_grid_districts, session=session)

# export grid to csv-files
try:
    nd.to_csv()
except:
    save_nd_to_pickle(nd, filename='ding0_grids_example.pkl')

```

Leonrtg, 2025-10-22:
Thank you much again! 

I double-checked all config files to ensure the host is correct:

- `~/.egoio/config.ini`  
- `~/.ding0/config/config_db_credentials.cfg`  
- `ding0/config/config_db_credentials.cfg`  

All have:
```
host = openenergyplatform.org
```

I included the following snippets in the example code and got the following outputs:

```python
import requests
r = requests.post(
    "https://openenergyplatform.org/api/v0/advanced/connection/open",
    headers={"Accept": "application/json"}
)
print("Probe /advanced/connection/open:", r.status_code, r.text[:120])
# --> Probe /advanced/connection/open: 200 {"content": {"connection_id": <...>}}

eng_ro = db.connection(readonly=True)
print("DBG ENGINE URL (ro):", eng_ro.url)
# --> DBG ENGINE URL (ro): postgresql+oedialect://openenergy-platform.org

engine = db.connection(readonly=False)
print("DBG ENGINE URL (rw):", engine.url)
# --> DBG ENGINE URL (rw): postgresql+oedialect://<username>:<token>@openenergyplatform.org:443/oedb  
# with <username> being my oep username and <token> the token I created on my profile 
```

I then found out that in egoio  (`/.../.venv/lib/python3.8/site-packages/egoio/tools/db.py`) the `openenergy-platform.org` link is hardcoded. 

```python
    host = input(
        'Enter value for `host` (default: openenergy-platform.org): ') or \
           'openenergy-platform.org'
    port = input('Enter value for `port` (default: 80): ') or '80'

... 

if readonly:
        conn = create_engine(
            "postgresql+oedialect://openenergy-platform.org")
    else:

```

I changed the links to `openenergyplatform.org ` and I change the number for the port (I set the port also to 443 in the config files): 

```
#port = input('Enter value for `port` (default: 80): ') or '80'
port = input('Enter value for `port` (default: 443): ') or '443'
```

Here the link to the first line: https://github.com/openego/ego.io/blob/2cb7bf082f63af25a83b8bb7011a5b9827b4048d/egoio/tools/db.py#L124

With the change I get the following outputs:

```python
# Probe /advanced/connection/open: 200 {"content": {"connection_id": <...>}}

# DBG ENGINE URL (ro): postgresql+oedialect://openenergyplatform.org  
# DBG ENGINE URL (rw): postgresql+oedialect://<username>:<token>@openenergyplatform.org:443/oedb  
```

Nevertheless, I still get the same error: 
[... 89 lines cut ...]
# database connection/ session
#engine = db.connection(readonly=False)
session = sessionmaker(bind=engine)()

# instantiate new ding0 network object
nd = NetworkDing0(name="network", session=session)

# choose MV Grid Districts to import
mv_grid_districts = [460]

# run DING0 on selected MV Grid District
nd.run_ding0(mv_grid_districts_no=mv_grid_districts)

# export grid to csv-files
try:
    nd.to_csv()
except:
    save_nd_to_pickle(nd, filename='ding0_grids_example.pkl')

```

Thanks again in advance for any help you provide me. 

Best,
Leon

closed 2026-10-30 by nesnoj